Breadstick-Innovations/Raspberry-Breadstick #12: Design panel with mousebites away from pins

mrangen, 2023-06-19:
Avoid v-scoring - depanelling will just lead to ripped pads at edge.
[image]

codeman256, 2023-06-30:
This was my current plan for panelling. 
[image]

mrangen, 2023-07-01:
That looks pretty good but I think we might need additional mouse-bites. I'd rather err on the side of too many than too few for the first panels.
[image]

mrangen, 2023-07-01:
We should also look into standard panel sizes and make sure that we're not designing a panel that's awkward or wasteful for typical assembly machines. Might be worth asking the manufacturer to give a penalization recommendation.

closed 2023-11-14 by codeman256