Breadstick-Innovations/Raspberry-Breadstick #24: Magic Wand of HarryPotter

JpEncausse, 2024-05-11:
Hello,
I was wondering if someone already did a HarryPotter Magic Wand to recognize gesture ?
- I know [there is some arduino code](https://github.com/boilingbrains/BaguetteMagiqueArduino/blob/main/CodesArduino/IMU_Classifier_Baguette_v2.ino) involving Tensorflow.
- Few year ago a company made a close source harrypotter wand.

It woudl be so cool to be able to recognize gesture and send HTTP Request for instance or special LED effect.

mrangen, 2024-05-12:
That's a very cool idea! I haven't played with machine learning on microcontrollers yet, but I bet it can be done. Hopefully someone in the community with experience will post some code!

JpEncausse, 2024-05-13:
I hope too :-) I used to play with PlatformIO on ESP-32 but I don't know CircuitPython and TensorFlow.
I assume each gesture could trigger a LED effect + HTTP request :-)

From Kano in 2018 : https://www.youtube.com/watch?v=wt9rK82l-io